buserror/simavr #419: Freerunning mode on attiny85 doesn't appear to ever trigger input/output updates

jckarter, 2020-12-12:
I was trying to simulate a small attiny85 program that implements a voltage-controlled PWM generator like so:

```
  .include "tn85def.inc"

  .cseg
  ; interrupt vectors
  .org 0
  rjmp start
  rjmp start
  rjmp start
  rjmp start
  rjmp start
  rjmp start
  rjmp start
  rjmp start
  .org ADCCaddr
  rjmp adc_ready
  rjmp start
  rjmp start
  rjmp start
  rjmp start
  rjmp start
  rjmp start

  ; entry point
  .org INT_VECTORS_SIZE
start:
  cli
  ; set up power mgmt
  ldi r16, 1<<SE ; enable sleep, idle mode
  out MCUCR, r16
  ldi r16, 1<<PRTIM1 | 1<<PRUSI ; shut down timer 1 and USI
  out PRR, r16
  ; pin config
  ldi r16, 1<<DDB1 ; output on pin 1
  out DDRB, r16
  ; pwm
  ldi r16, 1<<CS00 ; enable timer, no prescale
  out TCCR0B, r16
  ldi r16, 1<<COM0B1 | 1<<WGM01 | 1<<WGM00 ; clear on compare match, set at bottom on OC0B/PB1
  out TCCR0A, r16
  ldi r16, 0xbf
  out OCR0B, r16 ; initial duty cycle
  ; adc
  ldi r16, 1<<ADLAR | 1<<MUX0 ; Vcc reference, left adjust result, input channel ADC1/PB2
  out ADMUX, r16
  ldi r16, 0 ; auto trigger free running mode
  out ADCSRB, r16
  ldi r16, 1<<ADEN | 1<<ADATE | 1<<ADIE ; enable ADC, auto trigger, enable interrupt
  out ADCSRA, r16
  sei ; set interrupt
  sleep ; wait for adc interrupt

adc_ready:
  ; get rid of return address we don't need
  pop r16
  pop r16
  ; change PWM duty cycle based on reading
  in r16, ADCH
  out OCR0B, r16
  sei
  sleep
```

The program works on real hardware. However, I was having trouble trying to test it with simavr. Here is the simulator setup I've written:

```
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "avr_adc.h"
#include "avr_ioport.h"
#include "sim_avr.h"
#include "sim_elf.h"
#include "sim_hex.h"
#include "sim_vcd_file.h"

static void adc_out_trigger_notify(avr_irq_t *irq, uint32_t value, void *param) {
  avr_t *avr = param;

  printf(".\n");
  avr_raise_irq(avr_io_getirq(avr, AVR_IOCTL_ADC_GETIRQ, ADC_IRQ_ADC1),
                avr->cycle/200);
}

int main(int argc, char *argv[]) {
  elf_firmware_t f = {{0}};
  strcpy(f.mmcu, "attiny85");
  f.frequency = 1000000;
  ihex_chunk_p chunk = NULL;
  int cnt = read_ihex_chunks("pwm.hex", &chunk);
  if (cnt != 1) {
    fprintf(stderr, "unable to read single chunk from pwm.hex; got %d\n", cnt);
    exit(1);
  }
  f.flash = chunk[0].data;
  f.flashsize = chunk[0].size;
  f.flashbase = chunk[0].baseaddr;
  avr_t *avr = avr_make_mcu_by_name(f.mmcu);
  avr_init(avr);
  avr_load_firmware(avr, &f);

  avr_irq_register_notify(avr_io_getirq(avr, AVR_IOCTL_ADC_GETIRQ, ADC_IRQ_OUT_TRIGGER),
                          adc_out_trigger_notify,
                          avr);

  avr_vcd_t vcd_file;
  avr_vcd_init(avr, "pwmtest.vcd", &vcd_file, 100000 /* usec */);
  avr_vcd_add_signal(&vcd_file, 
                     avr_io_getirq(avr, AVR_IOCTL_IOPORT_GETIRQ('B'), IOPORT_IRQ_PIN1), 1 /* bits */ ,
                     "pwm");
  avr_vcd_start(&vcd_file);

  for (;;) {
    int state = avr_run(avr);
    if (state == cpu_Done || state == cpu_Crashed || avr->cycle >= 1000000)
      break;
  }

  avr_terminate(avr);
  avr_vcd_stop(&vcd_file);
  avr_vcd_close(&vcd_file);
}
```

However, it looks like the out trigger never fires, and so the ADCC interrupt never triggers. I had also tried modifying the AVR image to copy ADCH to OCR0B in a spinloop, but I also never see a change in the ADCH value. Am I holding it wrong? Does freerunning mode work on the tinyx5 cores?

gatk555, 2021-01-24:
The datasheet (section 7.1) says:

If an enabled interrupt occurs while the MCU is in a sleep mode, the MCU wakes up. The MCU is then halted for four cycles in addition to the start-up time, executes the interrupt routine, and resumes execution from the instruction following SLEEP.

So it seems your main program runs into the ISR, which then runs off the end.  That does not really explain what you are seeing, but fixing it would be a good next step.

Free-running mode can be seen working in simavr here (the test programs at the end):
https://github.com/gatk555/simavr/commit/eec2e5aae40d0831c385b6f8ff91fac466b4bb7b

The test will run without the simulator change, but will fail.

gatk555, 2021-01-24:
Setting ADSC may help.

jckarter, 2021-01-26:
Thanks @gatk555, I'll give that a try next time I get a chance to work on the project.

jckarter, 2021-01-30:
Yeah, setting ADSC looks like it did the trick. Thanks again @gatk555.

closed 2021-01-30 by jckarter