buserror/simavr #420: question about interrupts

jsmith173, 2020-12-14:
I have found that avr_clear_interrupt works on fifo items. avr_raise_interrupt works with table->vector items. When pending cleared it is cleared in the fifo item and at the next interrupt request I have got 'already raised' because pending is not cleared in table->vector items. Could somebody clarify it? Maybe I missed something while porting.

Example: simple timer0 example with interrupts

gatk555, 2021-01-24:
> I have got 'already raised' because ... 
I think it would help to include example code.

It is not clear to me what a "fifo item" is here.  Is it the "pending" fifo declared in sim_interrupts.h? It looks as if avr_clear_interrupt() uses only the vector structure in the MCU hardware description table and that is enough to prevent the interrupt,

The pending interrupt FIFO seems an odd feature, as the datasheets that I have looked at state that interrupts have priority
depending on their vector position.  If a pending interrupt is cleared, it seems that avr_service_interrupts() may exit without recognising other pending interrupts.  It will be caught on the next cycle of the run function, but does look like a bug.

There certainly seems to be a bug in avr_timer_write_pending() as it does not check for 1 bits before clearing interrupt flags.

gatk555, 2025-04-16:
Inactive discussion, no bug, so closed.

closed 2025-04-16 by gatk555