buserror/simavr #429: simavr via debian package manager

ManfredSteiner, 2021-01-29:
When I install simavr on a Debian (or Ubuntu) system via the package manager, the version 1.6+dfsg-3 is installed:  
[https://packages.debian.org/source/sid/simavr](https://packages.debian.org/source/sid/simavr)

The repository for these packages is located on:  
[https://salsa.debian.org/debian/simavr](https://salsa.debian.org/debian/simavr)

How does debian package version 1.6+dfsg-3 match versions (commits) on this repository?

edgar-bonet, 2021-01-29:
I tried this:

```shell
git remote add debian https://salsa.debian.org/debian/simavr.git
git fetch debian
git diff --compact-summary v1.6 debian/1.6+dfsg-3
```

It turns out that, compared to simavr v1.6, Debian's 1.6+dfsg-3,
 - adds the `debian` directory
 - removes `doc/manual`, `examples/board_simduino/ATmegaBOOT_168_atmega328.ihex` and `examples/shared/avr_twi_master.c`.

There are no other changes.

ManfredSteiner, 2021-01-31:
Thank you for response, nice "trick" to check differences.

closed 2021-01-31 by ManfredSteiner