buserror/simavr #434: Compiling with Qt 5.15.2 / MinGW 8.1.0 on Windows 10

xShadowEagle, 2021-02-07:
I am trying to compile *simavr* as part of my Qt project. My goal is to build a graphical front end and visualization of the different input and output signals of an Arduino (ATmega328).

I added the source code to my project and added all C source files to `SOURCES` and all Header files to the `HEADERS` section of the project definition. Now the project won't compile because the two following files include the Header file **`sim_core_decl.h`**.
```
$ grep -r -n --include "*.c" '#include "sim_core_decl.h"' .
./simavr/sim/run_avr.c:34:#include "sim_core_decl.h"
./simavr/sim/sim_avr.c:38:#include "sim_core_decl.h"
```

However I can't find this Header anywhere in the project directory.

```
$ find . -type f -name 'sim_core_decl.h'
```

Is the header somehow generated during compile time?

edgar-bonet, 2021-02-08:
Looks [like so][Makefile].

[Makefile]: https://github.com/buserror/simavr/blob/master/simavr/Makefile#L163

gatk555, 2021-02-09:
Yes, generated by simavr/Makefile (near the bottom).

> My goal is to build a graphical front end and visualization of the different input and output signals of an Arduino (ATmega328).

Something like this?

[image]

A work in progress, and Gtk rather than Qt, but there may be some scope for collaboration.

xShadowEagle, 2021-02-14:
> Something like this?
> 
> [image]
> 
> A work in progress, and Gtk rather than Qt, but there may be some scope for collaboration.

This looks pretty similar to what I want to build. In my project, the loaded program should be the exported binary from the Arduino IDE (hex format). Unfortunately I am required to develop under Windows and to build an executable for Windows. Because of this I'm having trouble with the required **libelf** library in Qt Creator. I can't seem to figure out where to get a precompiled version for Qt with MinGW. I tried to compile it myself but gave up after fixing seemingly endless dependency errors.

gatk555, 2021-02-24:
> I'm having trouble with the required libelf library in Qt Creator. 
I think it is a pretty unusual dependency, and have only seen it used once before.

Have you considered taking a hatchet to sim_elf.c and reducing the externally-accessible functions to stubs?  I think that is the only place libelf is used, and with .HEX files you will not need the ELF support.

If there is more to discuss on this, it may be a good idea to use the mailing list.  I do not look here very often.

xShadowEagle, 2021-02-27:
I got it to compile by installing **libelf** for MinGW-W64 in MSYS2 with the following command:

`pacman -S mingw-w64-x86_64-libelf`

And copying the following files into my project folder:

`/mingw64/lib/libelf.a`
`/mingw64/include/libelf/*`

> Have you considered taking a hatchet to sim_elf.c and reducing the externally-accessible functions to stubs? I think that is the only place libelf is used, and with .HEX files you will not need the ELF support.

This definitely seems more reasonable as I indeed do not need the ELF support.
Thanks for pointing this out. For now this issue can be closed as the compilation problem has been solved.

I will definitely get back to removing the **libelf** dependency when I'm optimizing my project some time later. If further problems arise with this I will open a new issue for this topic or contact the mailing list.

Best regards and thanks for everyone's help!

closed 2021-03-02 by xShadowEagle