buserror/simavr #437: Help with external connection

jsmith173, 2021-03-17:
Hi,

I'm trying to use simavr in an external gui. My problem with the external connections. For example: pin change interrupts. How simavr should be notified when an input changes? Which data structure should I update?

gatk555, 2021-03-17:
I recently did something similar and found the current manual to be less than ideal.  I wrote up what I learned as a "Getting Started" guide and I would be interested to know if you find it useful.  My initial draft is here:
https://www.dropbox.com/sh/q8k8b1a7ieou5t2/AABfGMWCaZWBmbe8tttjOkioa?dl=0
It does mention some software as being part of my fork that is not yet there, but it sounds as if that is not the part you will find useful.

jsmith173, 2021-03-31:
Thanks for the help file.

Now I'm studying the UART. I'm looking for how(where in the code) the TX pin is changing when UART communication in progress. I'm logged the avr_raise_irq with timestamp but I didn't found. TX pin is not changing in the UART communication?

gatk555, 2021-04-01:
My impression is that none of the communication peripherals are implemented
at the "hardware" level: they do not change or read I/O pins.  I think they
just simulate the internal behaviour of the peripheral to firmware and
provide IRQs to get the data in and out as bytes.  That is what I meant by
this in my document: "The serial communication periperals (UART, SPI, TWI)
are different, as transmissions do not modify pin state and reception does
not examine it. Instead, each peripheral has its own IRQs to send and
recieve bytes."

I have not looked at those parts of simavr much, as my interest is in
simulating firmware for control of analogue circuits.

jsmith173, 2021-04-01:
But this behaviour can change easily. Or not?
Anyway it means currently can't connect an external uart receiver.

gatk555, 2021-04-01:
I think that the simulation of communication peripherals to the pin level was seen as being a low priority.  So long as you are free to simulate the other end of the circuit in simavr's style it does not get you much, and simply calling avr_connect_irq() as is done in the examples is much, much simpler.  But if you want to use a pre-existing UART simulation it becomes important.

My guess is that it would be about 150 lines of code to implement the UART pins.  There are files avr_bitbang.* that look like an attempt to do pin-level SPI and they might be a partial model.  That code seems not used anywhere, perhaps because it is incomplete.

buserror, 2021-04-01:
Not only what @gatk555 says, but it would also slow down the simulation by a huge amount to try to actally keep all the pins in the state the hardware does, for very little gain. The peripherals outputs the "logical" interesting things out of the peripherals (bytes out of SPI, uart, i2c etc) but I don't see the point in trying to emulate a cycle accurate output pin.

lcgamboa, 2021-04-05:
@jsmith173 iIf your goal is to test Arduino codes using simavr for serial communication, it is better to use [PICSimLab](https://github.com/lcgamboa/picsimlab) directly.

Support for Arduino and interfaces with serial signal io (SPI, I2C and UART) are not objectives of the simavr project. PICSimLab uses a [modified version of simavr](https://github.com/lcgamboa/simavr), with support for I2C and SPI signals (master mode only) and Arduino (These changes would never be accepted in a pull request ). 
The uart signal generation part is in the [PICSimLab code itself](https://github.com/lcgamboa/picsimlab/blob/master/src/boards/bsim_simavr.cc). It will serve as a starting point if your goal is to make a simulator. 

Simavr is a great simulator, and i understand @buserror 's reasons for not adding detailed support to peripherals.

gatk555, 2025-04-08:
General question, was answered.

closed 2025-04-08 by gatk555