buserror/simavr #446: ihex extended address loading in additional chunks is incorrect

vintagepc, 2021-04-18:
There is a bug in read_ihex_chunks when loading a .hex file containing multiple segments with multiple "chunks"; the second binary after the first "end of file" is split into multiple chunks and has an incorrect starting offset:

Id      Size    Offset
0       250264  0
1       65536   196608
2       11028   65536

(Currently investigating, expect a PR w/fix soon)

gatk555, 2021-04-21:
The patch seems useful, but it is arguably not a bug, as references seem to say that an EOF record should be the last thing in the file.  Where did your multi-part one come from?

vintagepc, 2021-04-21:
It seems to be a quasi-common format supported by avrdude-slic3r and Prusa firmware .hex files where the first "chunk" is uploaded to the microcontroller and the subsequent chunk is written to an external flash chip. 

https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.9.3

PR #447 (2021-06-03): Fix #446 - bug in multipart ihex
Per #446 - there is a minor bug with multipart hex files because the segment is not reset to 0 when an "end of file" marker is reached. This causes the first segment of the next file to get an incorrect base address, and incorrectly breaks it into multiple chunks as well. 

Closes #446
commit 2d1045e5 (2021-06-03): Merge pull request #447 from vintagepc/446-fix-multipart-ihex

Fix #446 - bug in multipart ihex
commit 3a422e89 (2021-06-03): Fix #446 - bug in multipart ihex
closed 2021-06-03 by buserror