buserror/simavr #448: Delegate Maintenance

jhgorse, 2021-05-03:
Greetings.

It appears you might use some help with maintenance of the project.

What would it take for us to set up a little release process and delegate some repository maintainers to get some things off your stack and keep things moving here?

Cheers,
Joe

buserror, 2021-06-16:
All very nice, but I haven't had people contributing /steadily/ for a while. Some come and have a bash and a flurry then disappear. I might not be the most active maintainer in the world, but I'm still around, and I still follow up even if it's episodic.
It's episodic because I usually work on simavr when... I need it! Luckily enough I've come up with a new small project so that might liven things up for a bit.

vintagepc, 2021-07-21:
<-- Guilty as charged :rofl: 
> All very nice, but I haven't had people contributing /steadily/ for a while. Some come and have a bash and a flurry then disappear

I think part of that may also be a testament to the core design/setup; looking over the issues and PRs most of them seem to be corner cases or new features; at least in my experience with using SimAVR as a core for MK404... there's not a lot else that's needed beyond these longer term maintenance things like small bugfixes. The vast majority of features work very well. I've only had to dive into the innards a few times to explore unexpected behaviours or quirks. The only major thing I've found lacking recently is the missing timer modes. I had a go at it myself but just could not wrap my head around how that all works internally. :shrug:  (https://github.com/vintagepc/MK404/issues/202)

I look forward to seeing where this project goes and definitely don't want to see it go stale either :smile:

gatk555, 2025-04-02:
Closing this as I am a delegate and have done some maintenance.  And M.P. is not inactive.

closed 2025-04-02 by gatk555