buserror/simavr #456: WDIE is wronly cleared on interrupt in Interrupt Mode

WGH-, 2021-07-19:
`WDIE` should be only cleared in "Interrupt and System Reset Mode".

The fix is to add `&& avr_regbit_get(avr, p->wde)` condition.

https://github.com/buserror/simavr/blob/a56b550872906a971ac128002772d90c9e30377d/simavr/sim/avr_watchdog.c#L172-L188

I think I'll submit a PR (unless someone does that first) once I get access to hardware to double-check the tests on.

lcgamboa, 2021-12-10:
Hi @WGH- 

I have applied your fix suggestion to my simavr fork and now I can use WDT and Arduino_freeRTOS without any problems.
 I recommend you do a PR. Thanks for the info.

WGH-, 2021-12-10:
@lcgamboa just to clarify, Arduino_freeRTOS depends on the correct behaviour of `WDIE` auto-clearing, doesn't work correctly on simavr, and my suggested change fixes it?

itopaloglu83, 2021-12-10:
I was having issues with the watchdog timer interrupts and I'm glad to find this issue. Here's a simple example code for testing purposes. it uses the watchdog timer to blink an LED.

The watchdog timer interrupt only works once and then stops. I updated the interrupt clearing line with the following line of code as an interim solution.

Current: `WDTCSR |= (1 << WDIF);`
Temporary Solution: `WDTCSR |= (1 << WDIE) | (1 << WDIF);`

```c
#include <avr/io.h>
#include <avr/wdt.h>
#include <avr/interrupt.h>

ISR(WDT_vect)
{
    // Clear interrupt flag by writing 1 to it.
    WDTCSR |= (1 << WDIF);
    // Toggle LED.
    PORTB ^= _BV(PB5);
}

int main()
{
    // Set LED as output.
    DDRB |= _BV(DDB5);

    // Enable watchdog timer.
    WDTCSR |= (1 << WDCE) | (1 << WDE);
    WDTCSR = (1 << WDIE);

    // Toggle LED.
    PORTB ^= _BV(PB5);

    // Enable global interrupts.
    sei();

    // Loop forever.
    while (1)
    {
        _NOP();
    }
}
```

campfireman, 2025-01-22:
I had the same issue, @itopaloglu83 your workaround by setting `WDTCSR |= (1 << WDIE) | (1 << WDIF);` in the ISR worked. Thanks!

gatk555, 2025-04-07:
Believed fixed by d6ba83c1acf1.