buserror/simavr #457: avr_make_mcu_by_name: AVR '' not known

maxgerhardt, 2021-07-20:
After compiling simavr with MinGW with some caveats and modifications regarding the Makefile libelf and elf.h, I'm running into a problem where simavr can execute .hex files but not .elf files.

I use PlatformIO (`pio init -b uno`) to compile a simple firmware.elf and firmware.bin from the code

```cpp
#include <Arduino.h>

void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.println("Hello, world!");
  delay(1000);
}
```

When I use 

```
>simavr -t -m atmega328p -f 16000000 -v  C:\Users\Max\temp\simavr_test\.pio\build\uno\firmware.hex
Loaded 1 section of ihex
Load HEX flash 00000000, 1636
Hello, world!..
Hello, world!..
Hello, world!..
Hello, world!..
Hello, world!..
```

it works nicely, but when I run it on the elf file *and* in GDB (because otherwise no output?) I get

```
>gdb --args simavr -v  C:\Users\Max\temp\simavr_test\.pio\build\uno\firmware.elf
[...]
[New Thread 17300.0x2c54]
avr_make_mcu_by_name: AVR '' not known
C:\Users\Max\Desktop\simavr_installed\bin\simavr.exe: AVR '' not known
[Thread 17300.0x2508 exited with code 1]
```

Files are attached.

[uno.zip](https://github.com/buserror/simavr/files/6851741/uno.zip)

Any idea why it couldn't read the AVR from the ELF file?

ccrause, 2021-07-21:
If you don't specify the controller name on the command line, you need to embed it in the firmware. Use the [AVR_MCU](https://github.com/buserror/simavr/blob/a56b550872906a971ac128002772d90c9e30377d/examples/board_i2ctest/atmega1280_i2ctest.c#L28) macro to do this.

maxgerhardt, 2021-07-21:
Sadly I'm running into #458 then, but alright I see how it's supposed to work now.

maxgerhardt, 2021-07-21:
Closing since #458 is resolvable by creating a `.c` file with the macro needed for indentification.

closed 2021-07-21 by maxgerhardt