buserror/simavr #458: C99 designator 'string' outside

maxgerhardt, 2021-07-21:


ccrause, 2021-07-21:
Seems related to #285.  Issue seems to be related to compiling the macro output with C++.  There are numerous discussions on the web regarding this issue, see e.g.  [Why does C++11 not support designated initializer lists as C99?](https://stackoverflow.com/questions/18731707/why-does-c11-not-support-designated-initializer-lists-as-c99/29337570)

buserror, 2021-07-21:
You can sidestep the problem by creating a small .c that has the macros and link the .o with your C++. That way it is not compiled as C++. Ultimately this is not a simavr problem.
As to why C++ doesn't support it, well, that's one of the reason I stopped doing C++ many years ago. That syntax is fantastic and is used everywhere in simavr, all the 'cores' declarations are just giant designations that don't require glue code.

closed 2021-07-21 by buserror