buserror/simavr #464: build-example may fail linking with glu 9.0.2

zephvr, 2021-08-03:
### Summary

With Mesa OpenGL utility library (GLU) version 9.0.2 installed, running `make build-example` can fail with error like `undefined reference to symbol 'glEnd'` or `undefined reference to symbol 'glEnable'`.

A complete output can be found here [https://pastebin.com/ALfiRkw4](https://pastebin.com/ALfiRkw4)

### Details

This is an issue with glu version 9.0.2 and may affect different distributions (Archlinux in my case)

I added **MAY** because it only appear when building glu using meson and not make. Indeed while using meson the pkgconfig file generated is malformed and contains a `Requires.private: opengl` instead of `Requires: opengl` field. This lead to the command  `pkg-config --libs glu` to ommit the `-lGL` argument. Hence a linking error.

This issue have already been fix on the glu repo with this [commit](https://cgit.freedesktop.org/mesa/glu/commit/?id=225e00a14a9200f11253775a7c7778c1c29c8b23) We just have to wait for the next glu release.

### Quick fix

A temporary fix while an official version is release is to edit [Makefile.opengl](https://github.com/buserror/simavr/blob/master/examples/Makefile.opengl) and add a `--static` argument to `pkg-config --libs glu` commands

This gives:
`LDFLAGS 	+= ${shell pkg-config --libs --static glu} -lglut`

Berlinuxer, 2021-12-13:
Hello,
after i've edited the Makefile.opengl, i got the errormessages that ld cannot find -lGLU and -lglut. I am driving archlinux as well, and it is uptodate. Are there any other errors, or is it my linux, are there other packeges i need to install?

doppioandante, 2022-11-19:
Hi, I got the same problem with the `build-parts` part of the make process.
My fix was:
```
CPPFLAGS	+= ${shell pkg-config --cflags glu gl}
LDFLAGS 	+= ${shell pkg-config --libs glu gl} -lglut
```

closed 2025-04-07 by gatk555