buserror/simavr #474: Attiny85's PLOCK bit never sets (the PLL never locks)

packocrayons, 2022-01-31:
The following MWE reproduces the issue:
`
#include <avr/io.h>
#include <util/delay.h>

int main(){
        PLLCSR = (1 << PLLE);
        _delay_us(100);
        while (! (PLLCSR & (1 << PLOCK)));
        PLLCSR |= (1 << PCKE);
}
`
Running with simavr and avr-gdb (-ggdb -Og) and a breakpoint on line 8 (the PCKE line), never breaks. The emulator gets stuck on line 7 waiting for "hardware" to set the PLOCK bit to confirm that the PLL is locked

`avr-gcc -mmcu=attiny85 -Wall -DF_CPU=8000000 -Og -ggdb -o main main.c`

P.S. I would be happy to work on this issue, given a pointer on where to start to emulate the PLL

gatk555, 2022-02-02:
My guess is that the counter/timer code (avr_timer.*) is the natural home for that.  The data sheet documents PLLCSR in the counter/timer section.

[This link](https://www.avrfreaks.net/comment/793142#comment-793142) may be useful: other AVRs with PLL.

sergeyyarkov, 2025-03-28:
Same here.  Need to manually set PLOCK bit Ty run my code.

gatk555, 2025-04-02:
What would be an acceptable fix?  If internal source is used a simple time delay might be enough.  But one might require that the bit is never set if the external input has too much jitter.  And what will you do with the output? AFAIK clocking a timer from PLL is not implemented,

buserror, 2025-04-03:
I'd be happy with a timer; the idea is to make the firmware beleive it's working properly (and not lock up!) so toggling that bit after X cycles makes perfect sense to me.