buserror/simavr #484: Located sections are ignored by simavr

ManfredSteiner, 2022-04-12:
For a project with an Atmega324P I use special sections to ensure that some code is located on desired addresses. Works fine on real target, but these sections are not loaded proper in simavr.

**Any idea what is going wrong and how to solve this problem?**

To reproduce this problem I use an Ubuntu system with **avr-gcc (5.4.0)**, **avr-gdb (10.1.90.20210103-git)** and **simavr (1.6+dfsg-3)**.

1) **Create file `main.c`**

```C
#define ATT_SECTION_APP __attribute__((section(".app")))

int main () ATT_SECTION_APP;

int main () {
    asm("nop");
    return 0;
}
```

2) **Build project and locate section `.app` to `0x7100`**
```
avr-gcc -c -o main.o main.c
avr-gcc -o main.elf -Wl,-section-start=.app=0x7100  -mmcu=atmega324p main.o
```

3) **Check result in elf-file**

Disassemble elf-file with `avr-objdump -d main.elf`

```
Disassembly of section .app:

00007100 <main>:
    7100:	cf 93       	push	r28
    7102:	df 93       	push	r29
    7104:	cd b7       	in	r28, 0x3d	; 61
    7106:	de b7       	in	r29, 0x3e	; 62
    7108:	00 00       	nop
    710a:	80 e0       	ldi	r24, 0x00	; 0
    710c:	90 e0       	ldi	r25, 0x00	; 0
    710e:	df 91       	pop	r29
    7110:	cf 91       	pop	r28
    7112:	08 95       	ret
```

4) **Start simavr**
```
simavr -g -m atmega324p main.elf
```

5) **Start avr-gdb and show memory on location `0x7100`**

Start in another shell avr-gdb with `avr-gdb main.elf`.  
Then execute the following commands on gdb console:
* `target remote localhost:1234`
* `x/20b 0x7100`

As you can see, the memory locations starting with `0x7100` are cleared (= `0xff`) instead of showing the same content as in disassembly view. Symbols like `main` are shown as desired. 

```
(gdb) target remote localhost:1234
Remote debugging using localhost:1234
0x00000000 in __vectors ()
(gdb) x/20b 0x7100
0x7100 <main>:		0xff	0xff	0xff	0xff	0xff	0xff	0xff	0xff
0x7108 <main+8>:	0xff	0xff	0xff	0xff	0xff	0xff	0xff	0xff
0x7110 <main+16>:	0xff	0xff	0xff	0xff
```

gatk555, 2022-04-13:
A probable work-round would be to convert the ELF file to ihex and load that.  That does lose debug symbols.

The problem sounds similar to #462, with the same cause: simavr does not really follow the ELF spec.

ManfredSteiner, 2022-04-14:
Thank your for response. Your statement in [#462](https://github.com/buserror/simavr/issues/462) ...
> simavr loads only the .text, .data, .eeprom, .fuse and .lock sections and assumes the first two are contiguous. 

... seems to be true, which is definitely a failure in simavr.

In my project I build now in different ways. One variant for hardware target and one variant for simulation (without modification of default section configuration). That's my work-arround at the moment.

gatk555, 2024-11-01:
Should be fixed by PR #487

closed 2024-11-01 by gatk555