buserror/simavr #486: Unable to reproduce tests/atmega88_example.axf results

jimustafa, 2022-04-14:
Running
```
./simavr/run_avr tests/atmega88_example.axf
```
yields
```
Loaded 1722 .text at address 0x0
Loaded 114 .data
Loaded 4 .eeprom
Read from eeprom 0xdeadbeef -- should be 0xdeadbeef..
Read from eeprom 0xcafef00d -- should be 0xcafef00d..
```
which is inconsistent with the expected output documented in the README.

All the tests are passing...

I am on CentOS 8 and using `avr-gcc` installed through `arduino-cli`.

gatk555, 2022-04-20:
The README does not say that the program you build will produce that output, just "A program running with these instructions ... will display".  Evidently an older version of run_avr with a higher logging level was used.  Indeed, you can get closer by adding several "-v" options to the command.

So there seems to be no real issue here.

gatk555, 2022-04-20:
It has been logged before: #168 and #445.  The main part of #168 (empty VCD file) is fixed, probably by #440.

jimustafa, 2022-04-21:
Thanks for the references; yes, this issue (if even an issue) seems to be a duplicate.

So, this is just a case of the output shown in the README being out of date with the state of the main branch.

jamesrwaugh, 2025-05-03:
Good find @atluft 

Running into this myself, yeah, it appears the VCD change dumps do not work for **any** IO registers with write handlers.

1. `simavr` always attaches the VCD change to the IO mem IRQ
2. In @atluft 's analysis, **any** IO register with a side effect, aka write handler, skips the IO mem IRQ, thus no change is reported

If the change is made, the question is: Now we would be calling 2 IRQs, the IO one, and the peripheral specific one. There is a peripheral one in this case: the UART TX IRQ. Now, IRQs would be potentially duplicated.

atluft, 2025-05-03:
@jamesrwaugh , that approach is much better!
I can confirm changing `_call_register_irqs` as described by @jamesrwaugh produces VCD output that matches the readme.

Thank you for sharing and explaining the issue of duplicated IRQs. I see that now and prefer the solution you have presented.

gatk555, 2025-05-19:
The VCD output was fixed by e829f2bfd1, with related fixes in 5d42c8d3265.

closed 2025-05-19 by gatk555