buserror/simavr #489: Flush buffered standard steams

ivankravets, 2022-05-04:
Hi,

Firstly, a big THANK YOU for the amazing project! Glad to see that @PlatformIO users found it useful.

See initial https://github.com/platformio/platformio-core/issues/3965 posted by @adbancroft.

There is a problem with data buffering on the SimAVR/Windows side. Possible solutions:

- https://stackoverflow.com/questions/12450066/flushing-buffers-in-c
- https://stackoverflow.com/questions/7876660/how-to-turn-off-buffering-of-stdout-in-c

Please note that it works great on Unix.

# How to reproduce?

I've just created a simple Python script and tried to redirect standard streams to the PIPE. Subsequent reading of 1 byte hangs Python script. See code

`simavr.py`
```python
import sys
import subprocess

if __name__ == "__main__":
    with subprocess.Popen(
        sys.argv[1:],
        stdout=subprocess.PIPE,
        stderr=subprocess.PIPE,
        bufsize=0,
    ) as proc:
        print("reading...")
        proc.stdout.read(1)
        print("reading is done.")
```

Call SimAVR using Python bridge/script:

```
python simavr.py bin/simavr -m atmega328p -f 16000000L firmware.elf
```

gatk555, 2022-05-07:
You could suggest to your user to try stdbuf.  It is apparently included for Windows in msys2.  It should work if the program was built with MINGW, but not if MS build tools were used.

> Please note that it works great on Unix.
Only because of a bug (PR #490).  But stdbuf works after that is applied.

ivankravets, 2022-05-07:
Thanks for the quick response. Could you share somewhere **simavr** built with `msys2`? We will publish it to the registry. Thanks!

gatk555, 2022-05-07:
My understanding is that Msys2 is just MINGW with pre-built GNU utilities.  So you may already have it.  As far as I can see, there is no simavr documentation on building for Windows, but I got the impression that MINGW is the easy, perhaps the only, way.  I have not tried myself.

ivankravets, 2022-05-08:
@adbancroft, could you try this instruction https://github.com/buserror/simavr/blob/master/README.mingw ? Does this work for you?

gatk555, 2022-05-08:
I had forgotten those build instructions, as they are in a separate file.  But I suggest looking at [the fork by maxgerhardt](https://github.com/maxgerhardt/simavr) or merging PR #460 as the MINGW instructions are much newer.

ivankravets, 2022-05-09:
@gatk555 could you merge PR #460? The @maxgerhardt is from PlatformIO Community.

@maxgerhardt , could you help us with the new binaries for SimAVR? We can remove the warning from https://docs.platformio.org/en/latest/advanced/unit-testing/simulators/simavr.html 

Thanks in advance!

maxgerhardt, 2022-05-09:
I rebuilt the PlatformIO tool-simavr package for Win x64 with the latest branch plus my build fixes plus gatk555's logging fixes and execution on the commandline looks good. A simple Arduino firmware that prints 10 times per seconds executes nicley with it and with correct timing. I also see that the firmware output no goes to stdout instead of stderr. (Redirecting stderr to `nul` only shows firmware output and no simavr messages, which are now on stderr)

```
simavr.exe -m atmega328p .pio\build\uno\firmware.elf 2> nul
Iteration: 0..
Iteration: 1..
```

However, when PlatformIO calls into the binary (and simavr opens the GDB server), then
1. Timing seems to be about 10 times slower (`delay(100);` takes about 4 seconds, `delay(1000);` almost a minute instead of a second)
2. The firmware's serial output does not appear in the "Debug Console" in VSCode 
3. in general, simavr seems to postfix firmware output with "`..`" at the end, which mis-represents the true firmware output

lcgamboa, 2022-05-10:
>     1. Timing seems to be at least 10-40 times slower (`delay(100);` takes about 4 seconds, `delay(1000);` 
>          almost a minute instead of a second)
I solved this problem in PICSimLab using a modified [avr_callback_run_gdb](https://github.com/lcgamboa/picsimlab/blob/aa759162eb6334c7b246cc661570a279d8460596/src/boards/bsim_simavr.cc#L507), I believe it can be applied as PR in simavr, what's your opinion @gatk555 ?

edgar-bonet, 2022-05-10:
@maxgerhardt wrote:
> simavr seems to postfix firmware output with "`..`" at the end

This is meant to represent the `"\r\n"` line termination appended by `Serial.println()`. Simavr [replaces non-printing characters by `'.'`][replace] in the firmware output:

```c
p->stdio_out[p->stdio_len++] = v < ' ' ? '.' : v;
```

[replace]: ../blob/v1.7/simavr/sim/avr_uart.c#L275

gatk555, 2022-05-10:
A PR to fix the "real-time" behaviour with gdb sounds like a good idea to me.  You may run into a conflict with merged PR #482.  I did not worry about timing behaviour there, except to fix buzzing, as it seemed to be already broken.

It also looks a good idea to remove those trailing dots.  I think there are two places where they can occur, for UART and "console" output.

maxgerhardt, 2022-05-10:
I think simavr is more usefull if it outputs the firmware's intended output 1:1. Then you can even do things like having a firmware produce some binary output and pipe the simavr (= firmware output) into another program for processing or logging etc.

I'll ry and patch that in plus try and copy/paste the modified avr_callback_run_gdb.

Ah and, I've started setting up Msys2 MinGW64 + MinGW32 github action builds in https://github.com/maxgerhardt/simavr/actions/runs/2303871484, it already produces binaries.

gatk555, 2022-05-11:
I think the conversion to dots is very useful when running interactively and the firmware is misbehaving.  That is a common case, so "raw mode" should be a configurable option.

Windows binaries ought to be popular, but how will you advertise their availability?  A PR to the README would be best, but slow.  A Wiki entry perhaps.

maxgerhardt, 2022-05-11:
Best case, the CI build things get merged back and thus every commit will have downloadable artefacts for every major OS and architecture, plus the simavr release page will have the downloadable artefacts as well (see e.g. [here](https://github.com/earlephilhower/pico-quick-toolchain/releases)). Advertising my fork doesn't make sense, it has to go back into mainline.

gatk555, 2022-05-11:
The last round of merges was a month ago, with the one before 8 months previous.  So I think it would make sense to advertise the fork until the auto-build configuration is merged.

maxgerhardt, 2022-05-12:
I made a crude addition of a `-r / --raw` switch to simavr in commit https://github.com/maxgerhardt/simavr/commit/5616a101a1d26470d474147039bd306021043ec0, I now get the firmware output in its perfect raw form.

I tried to backport @lcgamboa fixes for execution speed in GDB server mode, which restored nominal execution speed indeed, but broke GDB's ability to pause the running code execution. Once I do `continue` in the GDB console, Ctrl+C does not respond anymore. This was working before the patch :(. 

Old
```
(gdb) continue
Continuing.
^C
Program received signal SIGINT, Interrupt.
0x000000ea in micros ()
```
After patch
```
(gdb) continue
Continuing.
^C^CThe target is not responding to interrupt requests.
Stop debugging it? (y or n) y
```
Any ideas there? (Binaries for each patch state are in [here](https://github.com/maxgerhardt/simavr/actions)).

gatk555, 2022-05-13:
It seems you merged a change from a fork that has not merged PR #482 into one that has, and that has partially reverted the PR.  Control-C handling was modified in the PR, so it is no surprise that it is now broken. In fact I am surprised gdb works at all.

If I understood correctly, you are trying to fix the timing of cpu_sleep() when running with gdb.  I hacked-up one of simavr's test firmwares to make a program that writes to the UART every two seconds.  With today's build of simavr upstream, I see no difference in timing with gdb.   I am running like this:

 ../simavr/run_avr --gdb atmega88_timer16.axf &
avr-gdb atmega88_timer16.axf -ex 'target remote :1234'
(gdb) continue

maxgerhardt, 2022-05-13:
> With today's build of simavr upstream, I see no difference in timing with gdb. I am running like this:

That's weird, I'm basically running it the same way you are, but with my firmware at https://github.com/maxgerhardt/pio-simavr-testing.

Can you test with for ATMega328P and the ELF file in [firmware.zip](https://github.com/buserror/simavr/files/8688718/firmware.zip)?  I do

```
simavr -m atmega328p -f 160000 firmware.elf
```
goes fast, and
```
simavr -m atmega328p -f 160000 -g firmware.elf
```
with `avr-gdb firmware.elf -ex "target remote :1234" -ex "continue"` goes really slow.

With exactly the current upstream repo I'm getting the ultra slow behavior. Print time is 10 times *per second*.

lcgamboa, 2022-05-13:
In my modification I only call the gdb_network_handler function when the simulation state is stopped and not in each iteration of the avr_callback_run_gdb function as in the original code, which is what causes the slowness problem. But I forgot to mention that I call the gdb_network_handler function every 100ms of simulation in another part of PICSimLab. I believe that using some timing mechanism it is possible to call the gdb_network_handler function to handle gdb signals directly in simavr.

gatk555, 2022-05-13:
Yes, that one is more than 10 times slower when run with gdb.  And my guess is also that making a system call for every AVR instruction is what slows it down.  The Arduino library is busy-waiting so it suffers badly, but mine spent almost all of its time in cpu_sleep() and executes very few instructions.  (I got 72Mb of simavr tracing output in a few seconds and there is no sleep instruction.)

gatk555, 2022-05-13:
Calling  gdb_network_handler() once every 100 instructions makes the speed reduction about 30%, which seems acceptable.
I tried to post a diff, but this site chewed it up as usual.

ivankravets, 2022-05-13:
@maxgerhardt tested on macOS - works great!

P.S: You are the monster with this CI build config => https://github.com/maxgerhardt/simavr/blob/combined_fixes_and_ci/.github/workflows/build.yml

I even forked your repo to keep it as a gist 🙏

ivankravets, 2022-05-13:
@maxgerhardt does it make sense to update [SimAVR in the PlatformIO Regsitry](https://registry.platformio.org/tools/platformio/tool-simavr) by your build? We could remove [this warning](https://docs.platformio.org/en/latest/advanced/unit-testing/simulators/simavr.html) before PlatformIO Core 6.0 which is planned to be released this Monday.

gatk555, 2022-05-13:
I tidied up my initial change and pushed to by fork as branch "gdb_burst".  I would make a PR but the only validation is firmware.zip, which seems meagre.  If anyone here has test material or wants to develop it further, that will be welcome.

maxgerhardt, 2022-05-15:
@ivankravets I have the CI building all PlatformIO `.tar.gz` archives now at https://github.com/maxgerhardt/simavr/releases, but the binaries don't seem to be working well with the GDB debugger / VSCode. Stepping over code only advances PC by two bytes, step-into in my firmware [throws an exception](ttps://user-images.githubusercontent.com/26485477/168458555-cc0ba246-7005-4797-9a23-122216b575d9.png) in string constructor code :/. Have to revert all C code back to mainline and test if behavior is the same + open issues.

ivankravets, 2022-05-15:
Do you mean to not hurry with updating the SimAVR package in the registry?

maxgerhardt, 2022-05-15:
Exactly. I'll update once I can verify that binaries are working well with PlatformIO+VSCode/GDB debugger.

maxgerhardt, 2022-07-05:
I haven't actively worked on this since the last post, had to do other things. The issue may still persist in the current mainline.

gatk555, 2022-07-14:
There are several useful changes that may come from this thread.  Going into some detail they are:

make stdout unbuffered, preferably controlled by an option;

remove translation of non-printable firmware output to dots, again should probably be optional;

drop, rather than translate, CR/LF characters in firmware output;

improve execution speed when the gdb server is active;

CI configuration so that new binaries are built when the source code changes.

The first three could be combined in a single PR.
One addition to CI configuration may be useful: separate binaries with tracing enabled.

Mine is now PR#499, faster running with gdb.

gatk555, 2026-08-28:
.Commit .[66eca78](https://github.com/buserror/simavr/commit/66eca78e57dde0f0878df1075a217b9e85213c00) added an option for raw output to stdout.

closed 2026-08-28 by gatk555