buserror/simavr #505: Impossible to load ihex eeprom

AndrolGenhald, 2022-09-20:
I'm trying to simulate the atmega328p with ihex firmware and eeprom using `simavr -m atmega328p -f 16000000L -ff firmware.hex -ee eeprom.hex -v -v -v`, but it's loading the eeprom as flash and crashing:

```
Loaded 1 section of ihex
Load HEX flash 00000000, 19148
Loaded 1 section of ihex
Load HEX flash 00000000, 1024
CORE: *** 0006: Invalid Opcode SP=08ff O=0009 
CORE: *** 000a: Invalid Opcode SP=08ff O=0007 
CORE: *** 03fc: Invalid Opcode SP=08ff O=0044 
avr_sadly_crashed
```

I see [here](https://github.com/buserror/simavr/blob/7003af00df0f89a38898896fe536c5f15ae4ef1a/simavr/sim/run_avr.c#L225) that `-ee` sets `loadBase` to `AVR_SEGMENT_OFFSET_EEPROM`, which is defined [here](https://github.com/buserror/simavr/blob/4d1750be7f430dec8a6c4ab06242d225a6a4250c/simavr/sim/sim_elf.h#L39) to be `0x00810000`, but in [`sim_setup_firmware`](https://github.com/buserror/simavr/blob/b3ea4f93e18fa5059f76060ff718dc544ca48009/simavr/sim/sim_hex.c#L221) the check for whether it's loading eeprom or flash is `chunk[ci].baseaddr < (1*1024*1024)`, which doesn't account for `loadBase`. The ihex format only allows 16 bits for the base address, so the max address is 65535, making it impossible to get simavr to load a file as eeprom instead of flash.

The [condition for eeprom](https://github.com/buserror/simavr/blob/b3ea4f93e18fa5059f76060ff718dc544ca48009/simavr/sim/sim_hex.c#L236) checks both the base address by itself and base address + `loadBase`, but since this happens after the flash condition which only checks base address, it can't be reached. It's unclear to me why `loadBase` isn't used in both conditions, and why the eeprom condition checks both with and without it.

AndrolGenhald, 2022-09-20:
I was able to work around the issue by adding an "Extended Linear Address" at the start of the eeporm file to indicate that the addressing should start at 0x100000:

```
:020000040010EA
```

This causes the eeprom to be loaded correctly, but as of yet I haven't figured out if it's possible to get simavr to save modified eeprom back to the file. It would be great to better document the eeprom functionality.

gatk555, 2022-09-20:
The loading code does not seem to have changed much since eeprom loading was introduced in 5d6254c8da54127276ecdd48069f00f19788693f.   Not checking loadBase before loading flash does seem a straightforward bug.   I think the reason that the eeprom address check is done both with and without loadBase is to accommodate two sorts of eeprom-containing ihex files: one that respects the convention used by avr-ld with the usual scripts (eeprom start address 0x81000) and one where eeprom contents are based at 0.  I guess you have the second sort.

A fix might also handle multiple non-overlapping chunks of both flash and eeprom, and non-increasing addresses in the ihex file.

An option to save eeprom in reloadable form also looks useful.

adbancroft, 2022-09-24:
Serendipity - I hit the same problem: PR to fix has been submitted.

PR #506 (2023-08-11): Include load base in all address checks when loading firmware
Fix #505
closed 2023-08-11 by buserror