buserror/simavr #510: IOT instruction (core dumped)

teackot, 2022-10-22:
A simple program with 2 delays:

```c
#ifndef F_CPU
#define F_CPU 9600000UL // 9.6 MHz clock speed
#endif

#include <stdbool.h>
#include <util/delay.h>
#include <avr/io.h>

int main() {
    DDRB |= 1;

    while (true) {
        PORTB |= 1;
        _delay_ms(300);
        PORTB &= ~1;
        _delay_ms(50);
    }
}
```

Launched with:

```
simavr -m attiny13 -f9600000 -g cmake-build-debug-avr/avr-hello-world
```

Gives:

```
Loaded 1204 .text at address 0x0
Loaded 0 .data
IOT instruction (core dumped)  simavr -m attiny13 -f9600000 -g cmake-build-debug-avr/avr-hello-world
```

However, if you leave only one _delay_ms or compile for another MCU (I tried ATMega328), program runs fine

edgar-bonet, 2022-10-22:
I could not reproduce this issue with simavr master (7003af00df0f89a38898896fe536c5f15ae4ef1a). I get instead:

```text
Loaded 82 bytes at 0
avr_gdb_init listening on port 1234
```

However, this line you showed looks suspicious:

> ```text
> Loaded 1204 .text at address 0x0
> ```

This cannot fit in an ATtiny13, and is also way too big for such a simple program. Did you forget to enable optimizations? If so, you should have seen this compiler message:

> **warning**: #warning "Compiler optimizations disabled; functions from &lt;util/delay.h&gt; won't work as designed" [-Wcpp]

If I attempt to load the unoptimized program (which is too large) with simavr, I get

```text
Loaded 1090 bytes at 0
Aborted
```

gatk555, 2022-10-23:
It works well for me also.  I can even see a lamp flashing:

[image]

If the size of the flash segment is the problem, adding option "-v" to the command will confirm it, as there is a warning message for the problem.  (There is a plausible PR to change WARNING to ERROR in the code.)  I think that is the right explanation, as "IOT instruction" is probably some shell's anachronistic way of saying "Aborted".  (SIGIOT and SIGABRT are the same thing.  I think it was the PDP-11 that actually had an IOT instruction!)

When I tried to compile without optimisation, the oversized flash was reported, with no output file:

In file included from attiny13_510.c:6:0:
/usr/lib/avr/include/util/delay.h:112:3: warning: #warning "Compiler optimizations disabled; functions from <util/delay.h> won't work as designed" [-Wcpp]

/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: attiny13_510.axf section `.text' will not fit in region `text'
collect2: error: ld returned 1 exit status

Building for an Atmega, then running on the Tiny13 fixed that, and I also saw "Aborted" and the message with "-v".

teackot, 2023-01-06:
Sorry for the late answer.

> Did you forget to enable optimizations

Yes, this was the problem. Enabling optimizations reduced the output size from 1204 B to 44 B and it now fits in ATtiny13

> If the size of the flash segment is the problem, adding option "-v" to the command will confirm it

Now with -v it shows an actual error message: `avr_loadcode(): Attempted to load code of size 1204 but flash size is only 1024.`

closed 2023-08-13 by buserror