buserror/simavr #530: Simulation freezes after completion of unity and next malloc

IngmarEckhardt, 2024-03-26:
Hey Community,

Maybe you want to look into the situation, I have troubles to debug it. When I run the test for the flash helper with unity in the simulator (simutron as wrapper) the simulation freezes at next malloc. I think malloc can't be the problem, I see now blocking things or loops there. 

Can't let simavr run as embedded gdb server, or with other words I lack the skills to add the virtual port to uart by my self. So I can't step with breakpoints through the code. 

It is in this branch, I linked the CMakeLists for the tests. It's happening in the isolated target for flash helper and the test runner too. After the tests for stdio(puts_pf) this don't happen. 

https://github.com/IngmarEckhardt/dwarfOS/blob/7-unit-tests-for-flash_helper/tests/CMakeLists.txt

IngmarEckhardt, 2024-03-28:
I updated this branch because added tests. Now this malloc freeze happens also inside the tests. Because its a branch under construction i added encapsulate printf around the failing malloc in the helper that its under test.
[image]
This is the situation at the serial terminal.
[image]

Will let this remote branch for sure at least two days in this condition, dont have so often a fire in the appartment :)).

gatk555, 2024-03-29:
I find this hard to understand, perhaps because your configuration is complicated.  For example, I have no idea what "unity" is here.

Two suggestions: have you tried enabling the tracing feature of simavr?  If the problem is an infinite loop in firmware, that will show it.  If not, it will show what the simulator is trying to do when it hangs.

Second: please post just the firmware that fails and any instructions needed to build it.  If some input is needed that can be recorded as a VCD file that can be played back in the simavr/run_avr program.  Then maybe someone will look at it.

buserror, 2024-03-29:
You could also try to attach gdb to the running process of simavr if it loops... and see what's going on. It's very likely the firmware is spinlooping somewhere, this leading simavr into 'freezing' as it continues emulating it...

IngmarEckhardt, 2024-03-29:
> I find this hard to understand, perhaps because your configuration is complicated. For example, I have no idea what "unity" is here.
> 
It is linked as a library in the CMakeLists? It is linked as a folder as github module? It is a lightweight testing library for c code.

> Two suggestions: have you tried enabling the tracing feature of simavr? If the problem is an infinite loop in firmware, that will show it. If not, it will show what the simulator is trying to do when it hangs.
> 

Thank you for this advice. I will try this.

> Second: please post just the firmware that fails and any instructions needed to build it. If some input is needed that can be recorded as a VCD file that can be played back in the simavr/run_avr program. Then maybe someone will look at it.

But i did this? The CMakeLists i linked contain every information to build the elf-file? It is the target with the separated test of the flash_helper? You can even watch the content of the whole flashmemory with a additional target?

IngmarEckhardt, 2024-03-29:
> You could also try to attach gdb to the running process of simavr if it loops... and see what's going on. It's very likely the firmware is spinlooping somewhere, this leading simavr into 'freezing' as it continues emulating it...

Thank you for this advice.

Found another strange behaviour, running simavr.exe as embedded gdb server, are you interested that i upload this in a different branch? But my next step is to use simavr in a wsl-ubuntu environment to get rid of problems that are maybe related to the windows build, maybe then such error dissappear.
[image]

IngmarEckhardt, 2024-03-31:
I am now in a linux environment with simavr installed apt-get packagemanager. Will look into the freezing malloc branche with this tracing feature (at first have to read how to do).

But also here i have this strange behaviour with the calculation, that its wrong with uint32_t as result. My CPU is slightly undervoltaged, never had any problems, maybe this is the first. This branch is still not uploaded, just playing around.
[image]

buserror, 2024-03-31:
You might be running out of stack. Try declaring half these variable 'static' and see if it helps

IngmarEckhardt, 2024-03-31:
> You might be running out of stack. Try declaring half these variable 'static' and see if it helps

This is with ELPM device, sry don't mentioned it, but I always name the run configurations recognizable, it's in the screen. it's a simulated atmega2560. This function getFreeMemory from my heaphelper that has the stack pointer gives me 7000k free memory between stack pointer and malloc heap.

gatk555, 2024-04-03:
Ingmar Eckhart,

Please read the first paragraph of the first comment on #531.

G,