buserror/simavr #531: atmega328p hd44780 assembly

EETagent, 2024-04-02:
I would like to ask the community for help.

Within the university labs, we are writing an AVR assembly in MPLAB (avrasm2) with a GUI plugin for the 1602 LCD Keypad. A shared plugin `printlib.inc` is used within all classes.

The 1602 uses hd44780 which the simavr module is already written for, so I decided to port the example code to the same schema.

I'm able to dump the characters to the lcd if I directly patch the initialization function in `printlib.inc` (it still crashes), but the external import ends up crashing immediately. It's functional in the Microchip propiertary emulator.

[image]

[image]

I have tried hex from both `avra` and `avrasm2`.

I'd love to get this working as an open minimal multi-platform solution instead of the manufacturer's crazy Netbeans IDE.

```asm
.include "asm/m328Pdef.inc"

.org 0x1000 ; <1>
.include "asm/printlib.inc"

.org 0
    jmp start

.org 0x100
start:
    call init_disp
    
    ldi r16, '0' 
    ldi r17, 0
    call show_char

end: jmp end
```

Repo to reproduce with all files

`make run`

https://github.com/EETagent/simavr-sap-experiment/tree/master/sap

gatk555, 2024-04-03:
This is like #530: here a problem that is described, but there are no good instructions to reproduce, just a link to a repository.  Is this a new trend? What I want to see is a small set of files (perhaps in tar format) that can be added to an existing simavr installation, compiled with the usual tools, and the problem reproduced.  Also, some clear instructions.  I imagine that most other people who might be able to help will want the same.  If you want someone to help you, you should first do some work to make it easy for them.

I was foolish, and clicked the link to the repository.  First thing I notice: the usual "Code" button for download is not there. It takes some action to remove that!  But it is easy to guess the URL to download with "git clone", so I made a copy.  Then tried "make run", from above, but there is no "run" target.  A simple "make" starts to build another copy of simavr, but then fails as not all pieces are here.  There is a subdirectory "sap" that looks like the application and firmware for reproduction.  And in here make works!  But when I run the emulator.elf file: "No hex / elf provided. Please provide a filename as an argument."  The firmware is not a default target of the Makefile!  If I look inside there is a target "asm".
but "make asm" yields: "avra: command not found".  It seems this firmware is to be built with a tool I do not have.  From Microchip?  And the code is not compiled by the command I usually use for AVR assembly source, "avr-gcc  -x assembler-with-cpp".  Now I came to my senses and stopped trying.

Enough grumble.  There may even be sufficient information to guess what is wrong: attempting to load into simavr a hex file made with Microchip tools.  It can be seen above that simavr loads most of the hex file at 0x2000 and rejects other parts.  Flash loading is supposed to start at 0 so the reset vector is defined.  My guess is that this firmware may have been built to work with a bootloader, but that has not been loaded.  Or the two small blocks that were rejected may contain bridging code.  Without the hex file, or the means to build it, who knows?

gatk555, 2024-04-03:
In fact the answer may be in your source code: the three .org statements match the chunks.  It seems that simavr's hex reading code lacks flexibility and has dropped parts of your program.  A simple fix may be to remove the first and third .org lines and move '.include "asm/printlib.inc"' after the reset vector.  Now there should be only one flash chunk.

EETagent, 2024-04-03:
Thanks for looking into my problem.

First, I'll clarify how to reproduce the repo. The Github link is just pointing to the /sap subfolder (Since it's a detail view , that's why you can't see the clone button, just move to repo root), which can just be copied to existing projects as you suggest. Otherwise the whole repo is provided for reproduction in a clean environment.

```sh
git clone https://github.com/EETagent/simavr-sap-experiment.git
cd simavr-sap-experiment/sap
```

if simavr lib is not installed globally, I updated root makefile, so you can build both library and /sap using make

avra is an open source compiler that is supposed to be compatible with reference 
Atmel's/Microchip's avrasm2. I don't distribute binaries in the introduction because it never looks legit. avrasm2 is only as a 32 bit exe for Windows (I can share the binary if you are able to run it), so few here could use it on Linux without Wine or VM. Fortunately avra is available in the official repositories for Debian / Ubuntu / Fedora / Arch and macOS via Homebrew with no other dependencies.

```sh
apt install avra
```

or from repo https://github.com/Ro5bert/avra with make

I could have shared the compiled hex files I'm trying to use straight away, you're right, so I'm attaching both the output from avra and the official avrasm2. They are same and included as a file in this comment (tar is not supported by GitHub as asset at the moment)

[avra_hex.zip](https://github.com/buserror/simavr/files/14852557/avra_hex.zip)
[avrasm2_hex.zip](https://github.com/buserror/simavr/files/14852558/avrasm2_hex.zip)

After running the make in the sap folder, the emulator will be in obj-***/emulator.elf. It takes a hex file as the first argument as seen in the screenshot.

**UPDATE:**

The suggestion in the second comment actually fixed the crash and now it works! Thanks a lot!

gatk555, 2025-04-16:
Should be fixed by cf0d394985f9.

closed 2025-04-16 by gatk555