buserror/simavr #538: atmega32 Timer0 CTC mode - Output Compare Interrupt always at TCNT0==0xFF

GeorgKorger, 2024-10-24:
Hi Michael!
I know, atmega32 is a pretty old MCU, but i have a lot of them, so i used them for my first experiments with simavr.
Unfortionally i stepped over an issue: Working with Timer0 in CTC mode does not bring the propper results in the simulation. The OVF Interrupt does not happen, when TCNT0 reaches OCR0, it happens when TCNT0 overflows at 0xFF. It seems, that the very first interrupt after reset happens on right time.
Same simulation with atmega1280 shows a propper result. So maybe all mcu's with only one Timer Compare Register show the issue?
To be quit sure, i programmed the firmware to a real world atmega32, who interrupts at the right time.

I used the Ubuntu 24.10 simavr package first, then i git cloned your master branch, but both showed the same issue.
Attachements: Source Code for mega32 and mega 1280 (zipped)
[puls8_Source_Codes.zip](https://github.com/user-attachments/files/17511415/puls8_Source_Codes.zip)

gtkwave output for both mcu's (screenshots):
[image]

 image of waveform from atmega32 - port B0:
[image]

I will try to debug the run_avr.elf code but i am not very sure if i am able to - i am only a "hobby programmer" ....

Greetings George

gatk555, 2024-10-31:
You could try my fork at https://github.com/gatk555/simavr.  It has several timer related changes including this one that looks somewhat promising:

    Author: ga <ga@oldell.fish>
    Date:   Wed Aug 10 18:55:02 2022 +0100

    Add support for basic phase-correct PWM modes (timer) with internal clock
    and some tests to verify.  Fix incorrect timer counts in CTC mode.
    In avr_ioport.c prevent double reporting of IOPORT_IRQ_PINx when a
    timer sets a PWM output pin, needed by test_atmega324a_timer16.c.

The changes were mostly aimed at adding the rest of the timer modes. but never requested for pull because still unfinished.

GeorgKorger, 2024-11-02:
Thank you, gatk555, but my issue is not addressed in this fork.
But i did a lot of tracing and debugging in the last days and it seems that the OCR Interrupt happens, but the timer is not resetted, so the next OCT Interrupt happens after 256 steps. It seems, that the atmega32 mcu even does not have simulated WGM0 bits...
I will try to search on.

gatk555, 2024-11-03:
I suggest trying to add the missing WGM bit definitions and value list using sim_megax4.h and similar files as a pattern.  That may be all that is needed.

GeorgKorger, 2024-11-03:
Yes, i found the right place to add the wgm bits and the wgm_op modes. I've attached a patch to change the cores/sim_megax.h. You can add this to your forks and branches.
[patch.zip](https://github.com/user-attachments/files/17611761/patch.zip)

closed 2024-11-03 by GeorgKorger