buserror/simavr #542: Mistake in commented example in avr_ioport.h

GeorgKorger, 2024-11-20:
I used the comment starting at line 58 of avr_ioport.h as template, and there was one missing parameter to avr_ioctl() . I created a patch for that file. Sorry for sending .zip File, but for some reason i am not allowed to upload .patch files.

Greeting, George

[avr_ioport_h_patch.zip](https://github.com/user-attachments/files/17836785/avr_ioport_h_patch.zip)

edgar-bonet, 2024-11-21:
Instead of uploading a .patch file, you can write the patch inline, preferably in [`git diff` format][git]. Put it inside a [fenced code block][md], with the `patch` language:

```patch
diff --git a/simavr/sim/avr_ioport.h b/simavr/sim/avr_ioport.h
index dba4407..35921ec 100644
--- a/simavr/sim/avr_ioport.h
+++ b/simavr/sim/avr_ioport.h
@@ -60,7 +60,7 @@ typedef struct avr_ioport_getirq_t {
  *
  * for (int i = 'A'; i <= 'F'; i++) {
  * 	avr_ioport_state_t state;
- * 	if (avr_ioctl(AVR_IOCTL_IOPORT_GETSTATE(i), &state) == 0)
+ * 	if (avr_ioctl(avr, AVR_IOCTL_IOPORT_GETSTATE(i), &state) == 0)
  * 		printf("PORT%c %02x DDR %02x PIN %02x\n",
  * 			state.name, state.port, state.ddr, state.pin);
  * }
```

The canonical “GitHub way” of proposing changes, though, is through a [pull request][pr]. This is in general more involved, but there is an easy shortcut suitable for simple changes like this one:

* navigate on GitHub to [the file you want to change][file]
* click on the pencil icon near the top right
* on the next page, click the green button “Fork this repository”
* make your edits to the file
* click on the green button “Commit changes...”
* when asked, enter a short commit message, e.g. “Fix comment in avr\_ioport.h”
* click on the green button “Propose changes”
* click on the green button “Create pull request”

It's simpler than it sounds: just follow the green buttons.

[git]: https://git-scm.com/docs/git-diff
[md]: https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks
[pr]: https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/about-pull-requests
[file]: https://github.com/buserror/simavr/blob/master/simavr/sim/avr_ioport.h

gatk555, 2024-11-21:
Nice guide, is there a way to make it sticky?

GeorgKorger, 2024-11-22:
Real nice guide - even i understood it 😉

closed 2024-11-22 by GeorgKorger

edgar-bonet, 2024-11-22:
@gatk555, @GeorgKorger: Thanks for your nice comments!

I am not aware of any means to make a comment sticky.