buserror/simavr #544: no data in .vcd after the first pin change

is-it-ayush, 2025-01-26:
Hi! I have the following code,
```C
#include "simavr/simavr/sim/avr/avr_mcu_section.h"
AVR_MCU(F_CPU, "atmega8a");
AVR_MCU_VCD_FILE("debug.vcd", 1000);
const struct avr_mmcu_vcd_trace_t _mytrace[] _MMCU_ = {
    // name of the trace, what variable to track, and what bit to track
    {AVR_MCU_VCD_SYMBOL("PB3"), .what = (void*)&PORTB, .mask = (1 << PB3)},
};

void pwm_constant_duty_cycle() {
  DDRB |= (1 << PB3); // set PB3 (OC2) as output
  TCCR2 |= (1 << WGM20) | (1 << COM21); // phase correct PWM mode; clear OC2 on compare match
  TCCR2 |= (1 << CS22) | (1 << CS21) | (1 << CS20); // 1024 prescaler
  OCR2 = 128; // 50% duty cycle
}

int main(void) {
  pwm_constant_duty_cycle();
  while(1) {}
}
```
To build the elf file for `simavr` simulation I run these,
```sh
# create object file
avr-gcc -mmcu=atmega8a -Os -Wall -ggdb -fno-omit-frame-pointer -fdiagnostics-show-option -std=c11 -I./include/ -I/usr/lib/avr/include -c main.c -o main.o

# create elf file
 avr-gcc -mmcu=atmega8a -Os -Wall -ggdb -fno-omit-frame-pointer -fdiagnostics-show-option -std=c11 -I./include/ -I/usr/lib/avr/include -o main.elf main.o
```
Now! I run simavr with `simavr -m atmega8 main.elf` but the generated trace (even if i run the simulation for 10 seconds) only contains data upto 1620ns (the moment PB3 pin changes) and then exits. I'm not sure what's wrong!

buserror, 2025-01-26:
The VCD will now show the PWM pin toggling; it only show the PORT/PIN/variable changes. You can display the PWM *registers* tho, if you want to track change, but the output PWM you probably assume should be there is not emulated.

is-it-ayush, 2025-01-26:
> The VCD will now show the PWM pin toggling; it only show the PORT/PIN/variable changes. You can display the PWM registers tho, if you want to track change, but the output PWM you probably assume should be there is not emulated.

Makes sense! Is there some place I can read up on what's not emulated so I can be aware of it? The readme said `Timers, 8 &16 (Normal, CTC and Fast PWM, the overflow interrupt too)` so I thought the generated pulses on OCx pins (which are part of "input/output changes") would be emulated too.

gatk555, 2025-05-22:
The code shown does not compile, and when that is fixed it does not run as atmega8a is not supported.  But the real problem is that Phase-Correct PWM is also not supported.  This version works (but should not!)

    #include "simavr/simavr/sim/avr/avr_mcu_section.h"
    #define F_CPU 8000000
    AVR_MCU(F_CPU, "atmega8");
    #include "avr/io.h"
    AVR_MCU_VCD_FILE("debug.vcd", 1000);
    const struct avr_mmcu_vcd_trace_t _mytrace[] _MMCU_ = {
    // name of the trace, what variable to track, and what bit to track
    {AVR_MCU_VCD_SYMBOL("PB3"), .what = (void*)&PORTB, .mask = (1 << PB3)},
    };

    void pwm_constant_duty_cycle() {
      DDRB |= (1 << PB3); // set PB3 (OC2) as output
      TCCR2 |= (1 << WGM21) | (1 << WGM20) | (1 << COM21); // fast PWM mode; clear OC2 on compare match
      TCCR2 |= (1 << CS22) | (1 << CS21) | (1 << CS20); // 1024 prescaler
      OCR2 = 128; // 50% duty cycle
    }
    
    int main(void) {
      pwm_constant_duty_cycle();
      while(1) {}
    }

It would be better to use AVR_MCU_VCD_PORT_PIN('B', PB3, "PB3") which should, and does, work.

closed 2025-05-22 by gatk555