buserror/simavr #552: Can't compile on MacOS Sonoma (Apple Silicon M1) from Homebrew

erosinnocenti, 2025-06-06:
Hello,
thanks to everyone for this great project.
I tried to install simavr on my Macbook Pro M1 Max (Sonoma 14.7.2) using `brew install --HEAD simavr` but unfortunately I got this compile error.
Am I missing something?
Thanks!

```
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C simavr RELEASE=0
/Applications/Xcode.app/Contents/Developer/usr/bin/make obj config
make[2]: Nothing to be done for `obj'.
make[2]: Nothing to be done for `config'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make libsimavr run_avr
make[2]: Nothing to be done for `libsimavr'.
LD obj-arm64-apple-darwin23.6.0/run_avr.elf
ld: warning: ignoring duplicate libraries: '-lelf', '-lm', '-lsimavr'
ld: warning: search path '/usr/local/opt/openssl@1.1/lib' not found
ld: warning: search path '/usr/local/opt/avr-gcc@12/lib' not found
ln -sf obj-arm64-apple-darwin23.6.0/run_avr.elf run_avr
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C tests RELEASE=0
AVR-CC atmega328pb_jsys.c
/opt/homebrew/bin/avr-gcc -Wall -gdwarf-2 -Os -std=gnu99 \
			-mmcu=atmega328pb \
			-DF_CPU=8000000 \
			-fno-inline-small-functions \
			-ffunction-sections -fdata-sections \
			-Wl,--relax,--gc-sections \
			-Wl,--undefined=_mmcu,--section-start=.mmcu=0x910000 \
			-I../simavr/sim/avr -I../simavr/sim/avr -I../../simavr/sim/avr \
			atmega328pb_jsys.c -o atmega328pb_jsys.axf
In file included from atmega328pb_jsys.c:10:
/opt/homebrew/Cellar/avr-gcc@9/9.4.0_1/avr/include/avr/io.h:581:6: warning: #warning "device type not defined" [-Wcpp]
  581 | #    warning "device type not defined"
      |      ^~~~~~~
In file included from atmega328pb_jsys.c:12:
/opt/homebrew/Cellar/avr-gcc@9/9.4.0_1/avr/include/avr/sleep.h:234:6: error: #error "No SLEEP mode defined for this device."
  234 |     #error "No SLEEP mode defined for this device."
      |      ^~~~~
atmega328pb_jsys.c: In function 'main':
atmega328pb_jsys.c:23:2: error: 'TCCR0A' undeclared (first use in this function)
   23 |  TCCR0A |= (1 << WGM01);                   // Configure timer 0 for CTC mode
      |  ^~~~~~
atmega328pb_jsys.c:23:2: note: each undeclared identifier is reported only once for each function it appears in
atmega328pb_jsys.c:23:18: error: 'WGM01' undeclared (first use in this function)
   23 |  TCCR0A |= (1 << WGM01);                   // Configure timer 0 for CTC mode
      |                  ^~~~~
atmega328pb_jsys.c:24:2: error: 'OCR0A' undeclared (first use in this function)
   24 |  OCR0A   = 0xAA;                           // CTC compare value
      |  ^~~~~
atmega328pb_jsys.c:26:2: error: 'TCCR0B' undeclared (first use in this function)
   26 |  TCCR0B |= (1 << CS00) | (1 << CS01);      // Start timer: clk/64
      |  ^~~~~~
atmega328pb_jsys.c:26:18: error: 'CS00' undeclared (first use in this function)
   26 |  TCCR0B |= (1 << CS00) | (1 << CS01);      // Start timer: clk/64
      |                  ^~~~
atmega328pb_jsys.c:26:32: error: 'CS01' undeclared (first use in this function)
   26 |  TCCR0B |= (1 << CS00) | (1 << CS01);      // Start timer: clk/64
      |                                ^~~~
atmega328pb_jsys.c:30:11: error: 'TIFR0' undeclared (first use in this function)
   30 |   while ((TIFR0 & (1 << OCF0A)) == 0) {
      |           ^~~~~
atmega328pb_jsys.c:30:25: error: 'OCF0A' undeclared (first use in this function)
   30 |   while ((TIFR0 & (1 << OCF0A)) == 0) {
      |                         ^~~~~
make[1]: *** [atmega328pb_jsys.axf] Error 1
make: *** [build-tests] Error 2
```

gatk555, 2025-06-06:
Strange.  My first guess is that your more recent compiler has somehow forgotten the Mega328PB, even though it seems to be the latest 328 variant.,  What does "avr-gcc --help-target" show? (But avr-gcc 7.3.0 does not list it, but still accepts "-mmcu=atmega328pb"!)

Another possibility is that it is somehow caused by using Apple's version of make.  The CI build (Github Actions) installs make and other packages with Homebrew before starting.  But is is X86, perhaps an ARM one should be added.

    HOMEBREW_NO_INSTALL_FROM_API=1 brew install make libelf freeglut patchelf 
    HOMEBREW_NO_INSTALL_FROM_API=1 brew tap osx-cross/avr
    HOMEBREW_NO_INSTALL_FROM_API=1 brew install avr-gcc@9 avr-binutils

If it is just the one file, deleting

    tests/*328pb*.c

will get past it with little loss, but I suspect there may be more problems beyond.

erosinnocenti, 2025-06-07:
Hi, thanks for you reply.

Avr-gcc version is `avr-gcc (Homebrew AVR GCC 9.4.0_1) 9.4.0`.
Here is the output of `avr-gcc --target-help`:
```
Known MCU names:
  avr1 avr2 avr25 avr3 avr31 avr35 avr4 avr5 avr51 avr6 avrxmega1
  avrxmega2 avrxmega3 avrxmega4 avrxmega5 avrxmega6 avrxmega7 avrtiny
  at90s1200 attiny11 attiny12 attiny15 attiny28 at90s2313 at90s2323
  at90s2333 at90s2343 attiny22 attiny26 at90s4414 at90s4433 at90s4434
  at90s8515 at90c8534 at90s8535 ata5272 attiny13 attiny13a attiny2313
  attiny2313a attiny24 attiny24a attiny4313 attiny44 attiny44a attiny84
  attiny84a attiny25 attiny45 attiny85 attiny261 attiny261a attiny461
  attiny461a attiny861 attiny861a attiny87 attiny43u attiny48 attiny88
  attiny828 at86rf401 at43usb355 at76c711 atmega103 at43usb320 attiny167
  at90usb82 at90usb162 ata5505 atmega8u2 atmega16u2 atmega32u2 attiny1634
  atmega8 ata6289 atmega8a ata6285 ata6286 atmega48 atmega48a atmega48pa
  atmega48p atmega88 atmega88a atmega88p atmega88pa atmega8515 atmega8535
  atmega8hva at90pwm1 at90pwm2 at90pwm2b at90pwm3 at90pwm3b at90pwm81
  at90pwm161 ata5790 ata5795 atmega16 atmega16a atmega161 atmega162
  atmega163 atmega164a atmega164p atmega164pa atmega165 atmega165a
  atmega165p atmega165pa atmega168 atmega168a atmega168p atmega168pa
  atmega169 atmega169a atmega169p atmega169pa atmega32 atmega32a atmega323
  atmega324a atmega324p atmega324pa atmega325 atmega325a atmega325p
  atmega325pa atmega3250 atmega3250a atmega3250p atmega3250pa atmega328
  atmega328p atmega329 atmega329a atmega329p atmega329pa atmega3290
  atmega3290a atmega3290p atmega3290pa atmega406 atmega64rfr2
  atmega644rfr2 atmega64 atmega64a atmega640 atmega644 atmega644a
  atmega644p atmega644pa atmega645 atmega645a atmega645p atmega649
  atmega649a atmega649p atmega6450 atmega6450a atmega6450p atmega6490
  atmega6490a atmega6490p atmega64rfr2 atmega644rfr2 atmega16hva
  atmega16hva2 atmega16hvb atmega16hvbrevb atmega32hvb atmega32hvbrevb
  atmega64hve at90can32 at90can64 at90pwm161 at90pwm216 at90pwm316
  atmega32c1 atmega64c1 atmega16m1 atmega32m1 atmega64m1 atmega16u4
  atmega32u4 atmega32u6 at90usb646 at90usb647 at90scr100 at94k m3000
  atmega128 atmega128a atmega1280 atmega1281 atmega1284 atmega1284p
  atmega128rfa1 atmega128rfr2 atmega1284rfr2 at90can128 at90usb1286
  at90usb1287 atmega2560 atmega2561 atmega256rfr2 atmega2564rfr2
  atxmega16a4 atxmega16a4u atxmega16c4 atxmega16d4 atxmega32a4
  atxmega32a4u atxmega32c4 atxmega32d4 atxmega32e5 atxmega16e5 atxmega8e5
  atxmega32x1 attiny212 attiny214 attiny412 attiny414 attiny416 attiny417
  attiny814 attiny816 attiny817 attiny1614 attiny1616 attiny1617
  attiny3214 attiny3216 attiny3217 atxmega64a3 atxmega64a3u atxmega64a4u
  atxmega64b1 atxmega64b3 atxmega64c3 atxmega64d3 atxmega64d4 atxmega64a1
  atxmega64a1u atxmega128a3 atxmega128a3u atxmega128b1 atxmega128b3
  atxmega128c3 atxmega128d3 atxmega128d4 atxmega192a3 atxmega192a3u
  atxmega192c3 atxmega192d3 atxmega256a3 atxmega256a3u atxmega256a3b
  atxmega256a3bu atxmega256c3 atxmega256d3 atxmega384c3 atxmega384d3
  atxmega128a1 atxmega128a1u atxmega128a4u attiny4 attiny5 attiny9
  attiny10 attiny20 attiny40
```

In fact, `atmega328pb` is missing.  
As you suggested, I deleted `tests/*328pb*.c` and it compiled fine.

closed 2025-09-19 by gatk555